avr-rust/ruduino #1: Automatically optimised multi-pin access

gergoerdi, 2017-03-13:
This feature request is for something like [Marangisto's AVR library](https://github.com/marangisto/AVR/blob/master/AVR/Pins.h) in C++, i.e. something that allows bunching of pins in such a way that reads/writes are optimized to take advantage of shared registers. In other words, if I bunch up the pins `PINB0`, `PINB1`, `PINB2`, `PINB3`, `PINC4`, `PINC2`, `PINB6`, `PINC0` into an 8-bit value, I should be able to read/write this 8-bit value with just two reads/writes (one for `PORTB` and one for `PORTC`). 

If you compile [this example](https://github.com/marangisto/AVR/blob/master/TestBits/Main.cpp) you will see that `bits::write(i++)` compiles to a single write to `PORTC`.

dcormier, 2020-07-31:
It looks like this functionality exists. This is from an example in the readme:
```
    let prev_value = read_volatile(PORTB);
    write_volatile(PORTB, prev_value ^ PINB5);
```

stappersg, 2022-06-08:
Breaking five years silence by closing this issue ...

closed 2022-06-08 by stappersg