avr-rust/ruduino #2: contributing other boards

dsvensson, 2017-06-25:
I just tried out this framework on the leonardo/promicro/atmega32u4 and had to change some addresses around to get for example serial working, but I guess the rest need some tweaking too. Do you have any nice idea on how to have different boards in the same code base?

dylanmckay, 2017-07-01:
> I just tried out this framework on the leonardo/promicro/atmega32u4 and had to change some addresses around to get for example serial working

What addresses were these? Did you have to run a custom linking command?

dylanmckay, 2017-07-01:
AVR-LLVM itself supports hundreds of different AVR microcontrollers, so the problem is likely that Rust needs to pass extra arguments to the linker.

dsvensson, 2017-07-02:
UDR0, UBRR0H, UBRR0L, UCSR0C, UCSR0B, UCSR0A etc

dylanmckay, 2017-07-02:
I'm getting my repositories mixed up haha, cc @shepmaster

shepmaster, 2017-07-05:
> Do you have any nice idea on how to have different boards in the same code base?

Yeah, that's a big question for this library!

My *hope* was that basically the [giant list of registers](https://github.com/avr-rust/arduino/blob/a09e4ca62efb561ad58544d649f2893eec895ea6/src/lib.rs#L38-L139) could be specified for each chip, perhaps in a module. That should work nowadays. 

The hard part comes with selecting the appropriate chipset. My first idea would be to have a feature flag that switches them around (and maybe enables certain parts like what serial features are available).

@dsvensson Could you maybe publish your fork somewhere so we could see what kind of diff you had to make? That might help guide what kind of swapability we need to enable.

dsvensson, 2017-07-07:
Going sligthly off-topic... I have very little experience in this area, but wouldn't it make sense to try to align this to the route @japaric is taking with embedded-hal etc, or is his arm efforts just too far away? Thinking otherwise it would be nice to get some cross pollination?

shepmaster, 2017-07-08:
> wouldn't it make sense to try to align this to the route @japaric is taking with embedded-hal etc, or is his arm efforts just too far away? Thinking otherwise it would be nice to get some cross pollination?

I'm always of two minds on this kind of thing. Having two parallel approaches has the potential for independent discovery of useful ideas. It also has the downside of duplicated effort for the common stuff.

For example, back in ye olde days, there was [zinc.rs](https://zinc.rs/) which offered one way of this type of abstraction. I haven't even looked at the ARM HAL stuff, so I don't know if it's the same idea as zinc or if they started from the same place.

However, @japaric is a smart embedded programmer, so I'd appreciate any input from them!

dsvensson, 2017-07-09:
Found a git repo with the datasheets in XML format, so here's a generated set of registers (and the script). Up for debate what the names should be.

http://7b014214dde8b39f.paste.se/

shepmaster, 2017-07-09:
Thanks. Would you mind running the script for the  atmega328p as well? That way I can compare it to the checked-in version.

<details>
<summary>Saved script for reference</summary>

```python
import xml.etree.ElementTree as ET
import argparse
import math

def bits(n):
    while n:
        b = n & (~n+1)
        yield int(round(math.log(b, 2)))
        n ^= b

def gen_bitfields(name, mask):
    fields = [None] * 8
    for bit in bits(mask):
        fields[7 - bit] = '{}{}'.format(name, bit)
    return tuple(fields)

def get_bitfields(node):
    fields = [None] * 8
    for bitfield in node.findall('bitfield'):
        name = bitfield.attrib['name']
        mask = int(bitfield.attrib['mask'], 16)
        for i, bit in enumerate(bits(mask)):
            if bin(mask).count('1') > 1:
                fields[7 - bit] = '{}{}'.format(name, i)
            else:
                fields[7 - bit] = name
    return tuple(fields)

def get_registers(node):
    for group in node.findall('.//register-group'):
        for register in group.findall('register'):
            name = register.attrib['name']
            offset = int(register.attrib['offset'], 16)

            if 'mask' in register.attrib:
                # If mask defined, there are no named bitfields
                mask = int(register.attrib['mask'], 16)
                if mask > 0xff:
                    # Split between high and low registers
                    yield (offset + 1, name + 'H', gen_bitfields(name + 'H', mask & 0xff00))
                    yield (offset, name + 'L', gen_bitfields(name + 'L', mask & 0xff))
                else:
                    yield (offset, name, gen_bitfields(name, mask))
            else:
                yield (offset, name, get_bitfields(register))

if __name__ == '__main__':
    # Datasheets can be found at https://github.com/avrxml/AS6-Devices-XML
    parser = argparse.ArgumentParser()
    parser.add_argument('datasheet', help='Atmel XML Register Datasheet.')
    args = parser.parse_args()

    for offset, name, bitfields in get_registers(ET.parse(args.datasheet).getroot()):
        fields = ''.join('{:11}'.format((f or '-') + ',') for f in bitfields).rstrip(', ')
        print('register!(0x{:02X}, {:9} [{:87}]);'.format(offset, name + ',', fields))

"""
atmega32u4:
register!(0xF4, UEINT,    [-,         UEINT6,    UEINT5,    UEINT4,    UEINT3,    UEINT2,    UEINT1,    UEINT0    ]);
register!(0xF3, UEBCHX,   [-,         -,         -,         -,         -,         UEBCHX2,   UEBCHX1,   UEBCHX0   ]);
register!(0xF2, UEBCLX,   [UEBCLX7,   UEBCLX6,   UEBCLX5,   UEBCLX4,   UEBCLX3,   UEBCLX2,   UEBCLX1,   UEBCLX0   ]);
register!(0xF1, UEDATX,   [DAT7,      DAT6,      DAT5,      DAT4,      DAT3,      DAT2,      DAT1,      DAT0      ]);
register!(0xF0, UEIENX,   [FLERRE,    NAKINE,    -,         NAKOUTE,   RXSTPE,    RXOUTE,    STALLEDE,  TXINE     ]);
register!(0xEF, UESTA1X,  [-,         -,         -,         -,         -,         CTRLDIR,   CURRBK1,   CURRBK0   ]);
[... 128 lines cut ...]
register!(0x38, TIFR3,    [-,         -,         ICF3,      -,         OCF3C,     OCF3B,     OCF3A,     TOV3      ]);
register!(0x36, TIFR1,    [-,         -,         ICF1,      -,         OCF1C,     OCF1B,     OCF1A,     TOV1      ]);
register!(0x35, TIFR0,    [-,         -,         -,         -,         -,         OCF0B,     OCF0A,     TOV0      ]);
register!(0x31, PORTF,    [PORTF7,    PORTF6,    PORTF5,    PORTF4,    -,         -,         PORTF1,    PORTF0    ]);
register!(0x30, DDRF,     [DDRF7,     DDRF6,     DDRF5,     DDRF4,     -,         -,         DDRF1,     DDRF0     ]);
register!(0x2F, PINF,     [PINF7,     PINF6,     PINF5,     PINF4,     -,         -,         PINF1,     PINF0     ]);
register!(0x2E, PORTE,    [-,         PORTE6,    -,         -,         -,         PORTE2,    -,         -         ]);
register!(0x2D, DDRE,     [-,         DDRE6,     -,         -,         -,         DDRE2,     -,         -         ]);
register!(0x2C, PINE,     [-,         PINE6,     -,         -,         -,         PINE2,     -,         -         ]);
register!(0x2B, PORTD,    [PORTD7,    PORTD6,    PORTD5,    PORTD4,    PORTD3,    PORTD2,    PORTD1,    PORTD0    ]);
register!(0x2A, DDRD,     [DDRD7,     DDRD6,     DDRD5,     DDRD4,     DDRD3,     DDRD2,     DDRD1,     DDRD0     ]);
register!(0x29, PIND,     [PIND7,     PIND6,     PIND5,     PIND4,     PIND3,     PIND2,     PIND1,     PIND0     ]);
register!(0x28, PORTC,    [PORTC7,    PORTC6,    -,         -,         -,         -,         -,         -         ]);
register!(0x27, DDRC,     [DDRC7,     DDRC6,     -,         -,         -,         -,         -,         -         ]);
register!(0x26, PINC,     [PINC7,     PINC6,     -,         -,         -,         -,         -,         -         ]);
register!(0x25, PORTB,    [PORTB7,    PORTB6,    PORTB5,    PORTB4,    PORTB3,    PORTB2,    PORTB1,    PORTB0    ]);
register!(0x24, DDRB,     [DDRB7,     DDRB6,     DDRB5,     DDRB4,     DDRB3,     DDRB2,     DDRB1,     DDRB0     ]);
register!(0x23, PINB,     [PINB7,     PINB6,     PINB5,     PINB4,     PINB3,     PINB2,     PINB1,     PINB0     ]);
register!(0x02, EXTENDED, [-,         -,         -,         -,         HWBE,      BODLEVEL2, BODLEVEL1, BODLEVEL0 ]);
register!(0x01, HIGH,     [OCDEN,     JTAGEN,    SPIEN,     WDTON,     EESAVE,    BOOTSZ1,   BOOTSZ0,   BOOTRST   ]);
register!(0x00, LOW,      [CKDIV8,    CKOUT,     SUT_CKSEL5,SUT_CKSEL4,SUT_CKSEL3,SUT_CKSEL2,SUT_CKSEL1,SUT_CKSEL0]);
register!(0x00, LOCKBIT,  [-,         -,         BLB11,     BLB10,     BLB01,     BLB00,     LB1,       LB0       ]);
"""
```
</details>

dsvensson, 2017-07-09:
Heh, didn't know about collapsable parts in comments before! Here you go:

<details><summary>ATmega328p</summary>

```
$ python pyavr.py AS6-Devices-XML/ATmega328P.xml|sort -r
register!(0xC6, UDR0,     [UDR07,     UDR06,     UDR05,     UDR04,     UDR03,     UDR02,     UDR01,     UDR00     ]);
register!(0xC5, UBRR0H,   [-,         -,         -,         -,         UBRR0H11,  UBRR0H10,  UBRR0H9,   UBRR0H8   ]);
register!(0xC4, UBRR0L,   [UBRR0L7,   UBRR0L6,   UBRR0L5,   UBRR0L4,   UBRR0L3,   UBRR0L2,   UBRR0L1,   UBRR0L0   ]);
register!(0xC2, UCSR0C,   [UMSEL01,   UMSEL00,   UPM01,     UPM00,     USBS0,     UCSZ01,    UCSZ00,    UCPOL0    ]);
register!(0xC1, UCSR0B,   [RXCIE0,    TXCIE0,    UDRIE0,    RXEN0,     TXEN0,     UCSZ02,    RXB80,     TXB80     ]);
register!(0xC0, UCSR0A,   [RXC0,      TXC0,      UDRE0,     FE0,       DOR0,      UPE0,      U2X0,      MPCM0     ]);
register!(0xBD, TWAMR,    [TWAM6,     TWAM5,     TWAM4,     TWAM3,     TWAM2,     TWAM1,     TWAM0,     -         ]);
register!(0xBC, TWCR,     [TWINT,     TWEA,      TWSTA,     TWSTO,     TWWC,      TWEN,      -,         TWIE      ]);
register!(0xBB, TWDR,     [TWDR7,     TWDR6,     TWDR5,     TWDR4,     TWDR3,     TWDR2,     TWDR1,     TWDR0     ]);
register!(0xBA, TWAR,     [TWA6,      TWA5,      TWA4,      TWA3,      TWA2,      TWA1,      TWA0,      TWGCE     ]);
register!(0xB9, TWSR,     [TWS4,      TWS3,      TWS2,      TWS1,      TWS0,      -,         TWPS1,     TWPS0     ]);
register!(0xB8, TWBR,     [TWBR7,     TWBR6,     TWBR5,     TWBR4,     TWBR3,     TWBR2,     TWBR1,     TWBR0     ]);
register!(0xB6, ASSR,     [-,         EXCLK,     AS2,       TCN2UB,    OCR2AUB,   OCR2BUB,   TCR2AUB,   TCR2BUB   ]);
register!(0xB4, OCR2B,    [OCR2B7,    OCR2B6,    OCR2B5,    OCR2B4,    OCR2B3,    OCR2B2,    OCR2B1,    OCR2B0    ]);
register!(0xB3, OCR2A,    [OCR2A7,    OCR2A6,    OCR2A5,    OCR2A4,    OCR2A3,    OCR2A2,    OCR2A1,    OCR2A0    ]);
register!(0xB2, TCNT2,    [TCNT27,    TCNT26,    TCNT25,    TCNT24,    TCNT23,    TCNT22,    TCNT21,    TCNT20    ]);
register!(0xB1, TCCR2B,   [FOC2A,     FOC2B,     -,         -,         WGM22,     CS22,      CS21,      CS20      ]);
register!(0xB0, TCCR2A,   [COM2A1,    COM2A0,    COM2B1,    COM2B0,    -,         -,         WGM21,     WGM20     ]);
register!(0x8B, OCR1BH,   [OCR1BH15,  OCR1BH14,  OCR1BH13,  OCR1BH12,  OCR1BH11,  OCR1BH10,  OCR1BH9,   OCR1BH8   ]);
register!(0x8A, OCR1BL,   [OCR1BL7,   OCR1BL6,   OCR1BL5,   OCR1BL4,   OCR1BL3,   OCR1BL2,   OCR1BL1,   OCR1BL0   ]);
register!(0x89, OCR1AH,   [OCR1AH15,  OCR1AH14,  OCR1AH13,  OCR1AH12,  OCR1AH11,  OCR1AH10,  OCR1AH9,   OCR1AH8   ]);
register!(0x88, OCR1AL,   [OCR1AL7,   OCR1AL6,   OCR1AL5,   OCR1AL4,   OCR1AL3,   OCR1AL2,   OCR1AL1,   OCR1AL0   ]);
register!(0x87, ICR1H,    [ICR1H15,   ICR1H14,   ICR1H13,   ICR1H12,   ICR1H11,   ICR1H10,   ICR1H9,    ICR1H8    ]);
register!(0x86, ICR1L,    [ICR1L7,    ICR1L6,    ICR1L5,    ICR1L4,    ICR1L3,    ICR1L2,    ICR1L1,    ICR1L0    ]);
register!(0x85, TCNT1H,   [TCNT1H15,  TCNT1H14,  TCNT1H13,  TCNT1H12,  TCNT1H11,  TCNT1H10,  TCNT1H9,   TCNT1H8   ]);
register!(0x84, TCNT1L,   [TCNT1L7,   TCNT1L6,   TCNT1L5,   TCNT1L4,   TCNT1L3,   TCNT1L2,   TCNT1L1,   TCNT1L0   ]);
register!(0x82, TCCR1C,   [FOC1A,     FOC1B,     -,         -,         -,         -,         -,         -         ]);
register!(0x81, TCCR1B,   [ICNC1,     ICES1,     -,         WGM11,     WGM10,     CS12,      CS11,      CS10      ]);
register!(0x80, TCCR1A,   [COM1A1,    COM1A0,    COM1B1,    COM1B0,    -,         -,         WGM11,     WGM10     ]);
register!(0x7F, DIDR1,    [-,         -,         -,         -,         -,         -,         AIN1D,     AIN0D     ]);
register!(0x7E, DIDR0,    [-,         -,         ADC5D,     ADC4D,     ADC3D,     ADC2D,     ADC1D,     ADC0D     ]);
register!(0x7C, ADMUX,    [REFS1,     REFS0,     ADLAR,     -,         MUX3,      MUX2,      MUX1,      MUX0      ]);
register!(0x7B, ADCSRB,   [-,         ACME,      -,         -,         -,         ADTS2,     ADTS1,     ADTS0     ]);
register!(0x7A, ADCSRA,   [ADEN,      ADSC,      ADATE,     ADIF,      ADIE,      ADPS2,     ADPS1,     ADPS0     ]);
register!(0x79, ADCH,     [ADCH15,    ADCH14,    ADCH13,    ADCH12,    ADCH11,    ADCH10,    ADCH9,     ADCH8     ]);
register!(0x78, ADCL,     [ADCL7,     ADCL6,     ADCL5,     ADCL4,     ADCL3,     ADCL2,     ADCL1,     ADCL0     ]);
register!(0x70, TIMSK2,   [-,         -,         -,         -,         -,         OCIE2B,    OCIE2A,    TOIE2     ]);
register!(0x6F, TIMSK1,   [-,         -,         ICIE1,     -,         -,         OCIE1B,    OCIE1A,    TOIE1     ]);
register!(0x6E, TIMSK0,   [-,         -,         -,         -,         -,         OCIE0B,    OCIE0A,    TOIE0     ]);
register!(0x6D, PCMSK2,   [PCINT7,    PCINT6,    PCINT5,    PCINT4,    PCINT3,    PCINT2,    PCINT1,    PCINT0    ]);
register!(0x6C, PCMSK1,   [-,         PCINT6,    PCINT5,    PCINT4,    PCINT3,    PCINT2,    PCINT1,    PCINT0    ]);
register!(0x6B, PCMSK0,   [PCINT7,    PCINT6,    PCINT5,    PCINT4,    PCINT3,    PCINT2,    PCINT1,    PCINT0    ]);
register!(0x69, EICRA,    [-,         -,         -,         -,         ISC11,     ISC10,     ISC01,     ISC00     ]);
register!(0x68, PCICR,    [-,         -,         -,         -,         -,         PCIE2,     PCIE1,     PCIE0     ]);
register!(0x66, OSCCAL,   [OSCCAL7,   OSCCAL6,   OSCCAL5,   OSCCAL4,   OSCCAL3,   OSCCAL2,   OSCCAL1,   OSCCAL0   ]);
[... cut ...]
register!(0x23, PINB,     [PINB7,     PINB6,     PINB5,     PINB4,     PINB3,     PINB2,     PINB1,     PINB0     ]);
register!(0x02, EXTENDED, [-,         -,         -,         -,         -,         BODLEVEL2, BODLEVEL1, BODLEVEL0 ]);
register!(0x01, HIGH,     [RSTDISBL,  DWEN,      SPIEN,     WDTON,     EESAVE,    BOOTSZ1,   BOOTSZ0,   BOOTRST   ]);
register!(0x00, LOW,      [CKDIV8,    CKOUT,     SUT_CKSEL5,SUT_CKSEL4,SUT_CKSEL3,SUT_CKSEL2,SUT_CKSEL1,SUT_CKSEL0]);
register!(0x00, LOCKBIT,  [-,         -,         BLB11,     BLB10,     BLB01,     BLB00,     LB1,       LB0       ]);
```

</details>

dsvensson, 2017-07-10:
> Any idea about 16-bit register pairs or aliases?

I'm generating them as H/L, with bits from 0...15, here they are:

```rust
// atmega32u4
pub const ADC:    *mut u16 = ADCL    as *mut u16;
pub const EEAR:   *mut u16 = EEARL   as *mut u16;
pub const ICR1:   *mut u16 = ICR1L   as *mut u16;
pub const ICR3:   *mut u16 = ICR3L   as *mut u16;
pub const OCR1A:  *mut u16 = OCR1AL  as *mut u16;
pub const OCR1B:  *mut u16 = OCR1BL  as *mut u16;
pub const OCR1C:  *mut u16 = OCR1CL  as *mut u16;
pub const OCR3A:  *mut u16 = OCR3AL  as *mut u16;
pub const OCR3B:  *mut u16 = OCR3BL  as *mut u16;
pub const OCR3C:  *mut u16 = OCR3CL  as *mut u16;
pub const SP:     *mut u16 = SPL     as *mut u16;
pub const TCNT1:  *mut u16 = TCNT1L  as *mut u16;
pub const TCNT3:  *mut u16 = TCNT3L  as *mut u16;
pub const UBRR1:  *mut u16 = UBRR1L  as *mut u16;
pub const UDFNUM: *mut u16 = UDFNUML as *mut u16;

// atmega328p
pub const ADC:    *mut u16 = ADCL    as *mut u16;
pub const EEAR:   *mut u16 = EEARL   as *mut u16;
pub const ICR1:   *mut u16 = ICR1L   as *mut u16;
pub const OCR1A:  *mut u16 = OCR1AL  as *mut u16;
pub const OCR1B:  *mut u16 = OCR1BL  as *mut u16;
pub const SP:     *mut u16 = SPL     as *mut u16;
pub const TCNT1:  *mut u16 = TCNT1L  as *mut u16;
pub const UBRR0:  *mut u16 = UBRR0L  as *mut u16;
````

Restioson, 2017-08-10:
Side note, does this library have support for the Mega? It does have an increased number of pins compared to the Uno and other similar boards, as well as having (2 iirc) more serial ports, which run on certain pins.

shepmaster, 2017-08-25:
@dylanmckay any idea if [avrd](https://github.com/avr-rust/avrd) can be of use here?

Restioson, 2017-08-25:
AVRd does have the pins afaik. It could solve the problem.

shepmaster, 2017-08-25:
> AVRd does have the pins afaik

It appears it has *access* to the pins (`bitfield`), but isn't currently making use of them.

Restioson, 2017-08-25:
What do you mean? The README indicates that you can use them like you would using the arduino library: https://i.imgur.com/PSw3jbT.png

shepmaster, 2017-08-25:
> What do you mean? The README indicates that you can use them like you would using the arduino library:

Yes, but this library currently also [relies on the pins associated](https://github.com/avr-rust/arduino/blob/bdc6a303fdabb591f8e6f482fdfbbcc77df5ca84/src/lib.rs#L122)

```rust
register!(0x25, PORTB,  [PORTB7,  PORTB6,  PORTB5,  PORTB4,  PORTB3,  PORTB2,  PORTB1,  PORTB0 ]);
```

I don't see `PORTB7` in those docs.

dylanmckay, 2017-08-26:
Yeah, we definitely have access to the bitfields from the pack files.

At the moment, the library only uses them in [generated documentation](https://docs.rs/avrd/0.1.1/avrd/atmega328p/index.html#port-modules).

We can definitely use them to generate `PORT[A-Z][0-9]` constants.

dylanmckay, 2017-08-26:
Can confirm that AVRd has definitions for every register across every device.

shepmaster, 2017-08-26:
Just because it was in my head, I wonder what kinds of type safety and structuring we could employ. For example:

```rust
struct PORTB;

impl PORTB {
    const PORTB7: u8 = 1 << 0;
}

struct Register(*const u8);

impl From<PORTB> for Register {
    fn from(_: PORTB) -> Self { Register(0x20 as _) }
}
```

This would make each register its own type, and then we could choose to implement traits across them.

I did something similar locally:

```
pub trait Information {
    const DDR: *mut u8;
    const IO: *mut u8;
    const PIN: *mut u8;
}

pub struct B;

impl Information for B {
    const DDR: *mut u8 = ::DDRB;
    const IO: *mut u8 = ::PORTB;
    const PIN: *mut u8 = ::PINB;
}
```

dylanmckay, 2017-08-26:
I was thinking if we want the `avrd` library to do something like this.

I'd really like the library to be quite low level that other devices built atop of, but if we create a trait like this then we can auto generate it for all devices, rather than having to rebuild all of it for the Arduino library.

Another option might be to extract most of `avrd` to a library. With that, we could add it as a build dependency for Arduino and then use it to generate structures ourselves.

shepmaster, 2017-08-27:
> Another option might be to extract most of avrd to a library.

That seems like it might be the appropriate path. A rough idea would be that the `arduino` library would have only generic things like

```rust
pub trait Information {
    const DDR: *mut u8;
    const IO: *mut u8;
    const PIN: *mut u8;
}
```

The user uses avrd (or something built on top of it) in their own build script (insert hand-waving here), generating code like

```
pub struct B;

impl Information for B {
    const DDR: *mut u8 = ::DDRB;
    const IO: *mut u8 = ::PORTB;
    const PIN: *mut u8 = ::PINB;
}
```

Which accounts for their specific board (e.g. which `PORTx` are available). They can then use the combination of the two libraries, effectively dependency-injecting the concrete board information.

This could also allow for the user to choose a serial implementation for macros like `println!` to use.

A big question in my mind is exactly how abstract we can be. Aren't there some boards that have like... "half" of a PORT defined? Will things like that cause massive pain?

dylanmckay, 2017-08-29:
We could abstract it on the GPIO level rather than the port level.

The pack files do say the number of pins supported on each port.

Something like

```rust
trait Pin {
    const DDR: *mut u8;
    const PORT: *mut u8;
    const PIN: *mut u8;
    const MASK: u8;
}

struct PB5;

impl Pin for PB5 {
    const DDR: *mut u8 = ::DDRB;
    const PORT: *mut u8 = ::PORTB;
    const PIN: *mut u8 = ::PINB;
    const MASK: u8 = 1<<5;
}
```

dylanmckay, 2017-08-29:
I remember there also being some sort of special IO register that `1` can be written to which will toggle the pin, regardless of its current state.

Because the XOR is done in hardware, it's faster than using `PORTB` to toggle.

If we abstract this on the pin level, we can write a `toggle` method to do this optimisation transparently.

shepmaster, 2017-08-29:
> I remember there also being some sort of special IO register that 1 can be written to which will toggle the pin, regardless of its current state.

Yep, that was one of the things that tripped me up earlier this year, and why I asked you about coalescing 8 consecutive `sbi` instructions for the same register.

You can see my current set of changes [in the `ports` branch](https://github.com/avr-rust/arduino/commit/f6ae799db1357fcbdaf0943305c7e0794aeb7be2). TL;DR yeah, there's a `toggle` function.

dylanmckay, 2017-08-29:
I've got my own local branch as well, exposing a `Pin` trait that gets implementations generated per mcu.

I'm also creating a `HardwareSerial` trait, and if I can understand enough about timers, will hopefully do that too.

dylanmckay, 2017-08-30:
Here's my WIP PR avr-rust/arduino#4

Restioson, 2017-08-30:
Related: could this be done for serial ports?

Example gist: https://gist.github.com/Restioson/47976cc261a1a9b788114283e98f2308

dylanmckay, 2017-08-30:
Yeah, I've done it for SPI so far.

Restioson, 2017-08-30:
Great :smile: