avr-rust/ruduino #6: Write trait doesn't work for serial

clebi, 2017-11-23:
Hello,

I have tried to implement the Write trait for the Serial struct and I haven't succeed making the write_fmt work. <br>
Both write_str and write_char works well and I can't seem to understand why the write_fmt doesn't work. I don't have many experience with rust so maybe you can figure out what is not working. I though since write_str is working well and as write_fmt is implemented in Write trait, it should be working.

Here is the simple implementation I did : 

```
impl Write for Serial {
    fn write_str(&mut self, s: &str) -> core::fmt::Result {
        for &b in s.as_bytes() {
            transmit(b);
        }
        Ok(())
    }
}
```

Thanks in advance

shepmaster, 2017-11-23:
> I haven't succeed
> write_fmt doesn't work
> what is not working
> it should be working.

You never once have said what it means to "work" or "not work", so I'm not sure how anyone could hope to help.

dylanmckay, 2017-11-23:
I have a feeling that this may be related to the fact we have a custom
libcore with some function implementations commented out to appease the avr
backend.

My first thought would be to check if anything obvious is commented out in
the fmt module

shepmaster, 2017-11-23:
Good point. Probably a duplicate of avr-rust/libcore#3

clebi, 2017-11-23:
OK, sorry.

I meant when I use write_str, it outputs the str I passed as parameter. But when using write_fmt, there is no output at all.

I hope this is more clear.

shepmaster, 2017-11-23:
@clebi here's what's probably happening:

As part of avr-rust/libcore#3 we explicitly commented out the code that implements the formatting logic, ultimately because the LLVM infrastructure cannot handle it (avr-rust/rust#37). Someone needs to address the original LLVM issue before we can support *any* code that uses formatting.

clebi, 2017-11-23:
Thank you for your answers. I close it since it is a known problem.

closed 2017-11-23 by clebi

dylanmckay, 2017-11-24:
Thanks for filing this issue @clebi - this is a bit of a papercut.

I will add a "Caveats" section to the README so this is more obvious in the future. We can add to this list when we encounter more common problems.

lord-ne, 2021-08-23:
The trimmed-down libcore isn't necessary anymore, right? So this should work now?