avr-rust/ruduino #12: Unified interrupt handler interface

dylanmckay, 2018-11-07:
CC #9 

Our interrupt handler story isn't super great; we require users to write their own interrupt vector table (IVT) in assembly, and then they define a set of `pub unsafe extern "avr-interrupt" fn <interrupt handler symbol name>() { }` that the IVT points to.

[Here's](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo/blob/596ac4d69513c6701ee08b8e3f2d909f9ed2a2ca/interrupt_vector.S) an example program that installs interrupt handlers.

We should come up with a single, unified method of defining interrupt handlers. All AVR-Rust programs should use the same interrupt handler function names, by convention. This is similar to AVR-GCC, which always uses the same function names, although GCC has much more abbreviated, hard-to-understand names. Let's be more descriptive.

dylanmckay, 2018-11-07:
For example, we could write an `avr-interrupts` crate or something to do this. It is probably not a good idea baking this into the avr-rust/rust compiler itself because that will make the upstream effort much more difficult.

dylanmckay, 2018-11-07:
Whatever method we come up with, we need to be able to guarantee that the interrupt vector table will be linked **before** any other object file in any AVR-Rust program. The IVT always sits at the very first byte of a raw binary AVR program.

shepmaster, 2018-11-07:
> we require users to write their own interrupt vector table (IVT) in assembly

I don't think this is strictly true if you are linking with the avr-gcc runtime glue. AIUI, in that case, that code provides the IVT and other routines like "copy static data from flash to memory". 

> `avr-interrupts` crate or something to do this

I agree. Theoretically, people will want to do their own route for everything (hello, me) and baking anything in will make them annoyed.

> the interrupt vector table will be linked **before** any other object file in any AVR-Rust program.

I think the safe way to do this is via a linker script. You can see [mine](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo/blob/596ac4d69513c6701ee08b8e3f2d909f9ed2a2ca/linker-script). The table is stored in a special section and then linked first in the code.

shepmaster, 2018-11-07:
A related aspect that we will need to deal with is model-specific aspects. For example, this linker script bakes in the amount of memory available. It also has a fixed set of interrupts which change depending on the model of processor.