avr-rust/ruduino #31: Make `Register::set` use `volatile_write`

MalteT, 2020-08-03:
I was writing a small programs that uses the internal EEPROM of the ATmega328P, when I noticed that no writing was occuring.
I was using
```rust
EECR::set(EECR::EEMPE);
EECR::set(EECR::EEPE);
```
to start writing to the EEPROM, but the memory was unchanged.

When I replaced the above code with the following:
```rust
EECR::write(EECR::EEMPE);
EECR::write(EECR::EEPE);
```
it worked like a charme.

I might be mistaken, but after digging through the `Register` code, I noticed that `Register::set` is not using volatile_write
https://github.com/avr-rust/ruduino/blob/302095dca22f63a160d44c1fdcdc0fdf1db4072d/src/register.rs#L53-L57
while `Register::unset`, `Register::write`, etc are doing so. Is there a particular reason for this, that I may not understand yet?
https://github.com/avr-rust/ruduino/blob/302095dca22f63a160d44c1fdcdc0fdf1db4072d/src/register.rs#L70-L74

Thanks!

laris, 2020-09-18:
I also find the bug and you need to change the code.

dylanmckay, 2020-10-29:
Indeed, every other method in this file was changed to use volatile memory operations, but this one method was fixed. I've given the file two passes over checking to see if there are any other nonvolatile memops on registers but this seems to be the only one. Now fixed.

Thanks for raising this!

I've released [v0.2.7](https://crates.io/crates/ruduino/0.2.7) which contains the fix for this issue.

closed 2020-10-29 by dylanmckay