avr-rust/ruduino #34: Docs.rs is complaining this won't compile

timbell87, 2020-11-24:
I can't seem to get documentation for any version newer than 0.1.2 because "docs.rs failed to build ruduino-0.2.0"

This seems to be the same issue as here: https://github.com/avr-rust/blink/issues/25, I did a quick test and was able to confirm build failure and was able to get it working using the same fix I suggested there (included below for your reference).

**Cargo.toml**
```
[profile.dev]
opt-level = "s"
debug-assertions = false

[profile.release]
opt-level = "s"
```

dylanmckay, 2021-01-27:
It looks like the docs.rs build is failing because the `build.rs` script generates the Rust source code files in `src/cores/*.rs` at build time, but docs.rs operates with a read only source directory and so it won't work.

Here's the error: https://docs.rs/crate/ruduino/0.2.7/builds/310297

The best way to work around this would be to pregenerate the cpu-specific core configuration files and commit them to version control, that way they do not need to be regenerated on each build on every machine.

dylanmckay, 2021-01-29:
Will be fixed by #36

dylanmckay, 2021-01-29:
Fixed: https://docs.rs/ruduino/0.3.0/ruduino/

closed 2021-01-29 by dylanmckay