avr-rust/ruduino #42: "out of registers" build error

michkrom, 2022-03-03:
avr-gcc 5.4
ruduino 0.3.2

   Compiling rustc-std-workspace-core v1.99.0 (/home/m/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
error: ran out of registers during register allocation

error: could not compile `core` due to previous erro

n8henrie, 2022-05-19:
I have had this error sometimes resolve by building with `--release`.

stappersg, 2022-05-21:
Also encountered in https://github.com/avr-rust/delay/issues/21

stappersg, 2022-06-08:
@michkrom please try again, I would like to known if _ruduino_ also compiles for you.
```text
$ AVR_CPU_FREQUENCY_HZ=8_000_000 cargo +nightly build --target ./avr-atmega328p.json -Z build-std=core --release
   Compiling core v0.0.0 (/home/gs0604/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling ruduino v0.4.0 (/home/gs0604/src/rust/RustAVR/ruduino)
   Compiling rustc-std-workspace-core v1.99.0 (/home/gs0604/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.73
   Compiling avr-config v2.0.1
   Compiling avr-std-stub v1.0.3
   Compiling avr_delay v0.4.0 (https://github.com/avr-rust/delay#cfe9848f)
    Finished release [optimized] target(s) in 6.30s
```